NordicSemiconductor/pc-nrfutil #260: python3.7:nrfutil keys generate private.key error

jiangdailun, 2019-10-10:
I use "nrfutil keys generate private.key",but there is a mistake
Traceback (most recent call last):
  File "/usr/local/bin/nrfutil", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/nordicsemi/__main__.py", line 434, in generate
    signer.gen_key(key_file)
  File "/usr/local/lib/python3.7/site-packages/nordicsemi/dfu/signing.py", line 79, in gen_key
    sk_file.write(self.sk.to_pem())
TypeError: write() argument must be str, not bytes

bihanssen, 2019-10-11:
We're not able to reproduce. Also the line number in main.py does not correspond with what's in master currently. Make sure that all code is up to date. Reopen if you still see the problem.

akamensky, 2020-03-07:
@bihanssen I have same exact error.

Python 3.7. nrfutil installed from pip as directed on Nordic website. Version 6.0.1.

I don't think just outright closing this issue without asking for any additional info is a good idea...

akamensky, 2020-03-07:
It appears that pypi.org has very different code versions for Python2 and Python3, even command line arguments are different...

Installing nrfutil for Python2 does work. Python3 it fails to generate key.

bihanssen, 2020-03-09:
@akamensky, I will reopen the issue so it can be looked further into.

Note, if you run pip install with python 2 (ref your latest comment) you will get nrfutil v5.2.0. Could that be the reason you are seeing very different code version?

akamensky, 2020-03-09:
Correct, I missed that, python2 pip installed 5.2.0 version.

I have not tried installing from sources directly.

bihanssen, 2020-04-24:
Unless more info is provided on this I will assume the problem is mixing of versions 5.2 and 6, which mark the change from python 2 to 3. 
Causes that can lead to the problem are that 
1: nrfutil 5.2 does not disallow installation with python 3 even though it does not support python 3, 
2: the ble-driver-py dependency does not support pip < v20, which can lead to unfortunate fallback to version 5.2.

kenr, 2023-01-17:
We do not have capacity to maintain and release new versions of this utility.
A new utility with the same name will cover this feature. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr