NordicSemiconductor/pc-nrfutil #265: AttributeError: 'int' object has no attribute 'encode' on MacOS Catalina

hecko, 2019-10-16:
Im hitting the same issue om MacOS Catalina with Python3 from brew and nrfutils installed as

`pip3 install --upgrade --force-reinstall  nrfutil --user`

This is the result:

`Marcels-MBP-511:dfu marcel$ /Users/marcel/Library/Python/3.7/bin/nrfutil pkg generate --hw-version 52 --sd-req 0xB8 --application-version 1 --application ../../FW/norman_5_firmware/pca10056e/s112/ses/Output/Release/Exe/ble_app_uart_pca10056e_s112.hex --key-file key.pem norman5_dfu_v1.zip
Traceback (most recent call last):
  File "/Users/marcel/Library/Python/3.7/bin/nrfutil", line 10, in <module>
    sys.exit(cli())
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/nordicsemi/__main__.py", line 815, in generate
    default_key = signer.load_key(key_file)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/signing.py", line 92, in load_key
    sk_hex = "".join(c.encode('hex') for c in self.sk.to_string())
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/signing.py", line 92, in <genexpr>
    sk_hex = "".join(c.encode('hex') for c in self.sk.to_string())
AttributeError: 'int' object has no attribute 'encode'`

Any thoughts?

_Originally posted by @hecko in https://github.com/NordicSemiconductor/pc-nrfutil/issues/153#issuecomment-542525929_

hecko, 2019-10-16:
OK, aparently according to the docs on this utility is only compatible with python2 so on mac do this: 

`easy_install pip
pip install nrfutil`

Make sure you run nrfutil using python2, not python3

closed 2019-10-16 by hecko

bihanssen, 2019-10-16:
Hi @hecko, 
that's correct, nrfutil (<=v5) is Python 2 only. 
The next version of nrfutil (v6) will be include python 3 support.