NordicSemiconductor/pc-nrfutil #275: Clarify Python 3 version requirement

mbolivar, 2019-10-22:
setup.py has different strings indicating that either Python 3.6 or 3.7 is the minimum supported version.

These should be aligned.

Also, please strongly consider supporting Python 3.6. This is the version which is provided with Ubuntu 18.04, the current LTS.

mbolivar, 2019-10-22:
cc @carlescufi as this is an NCS dependency

bihanssen, 2019-10-23:
Hi @mbolivar, 
the issue for us is that nrfutil has one dependency, pc-ble-driver-py, that does not work on Python 3.6. When trying to install nrfutil, that dependency would fail the installation. All other parts of nrfutil does support py3.6.

mbolivar, 2019-10-23:
> the issue for us is that nrfutil has one dependency, pc-ble-driver-py, that does not work on Python 3.6. 

I see, but that does not address this problem in the issue report:

> setup.py has different strings indicating that either Python 3.6 or 3.7 is the minimum supported version.
> These should be aligned.

> When trying to install nrfutil, that dependency would fail the installation. All other parts of nrfutil does support py3.6.

Is this a mandatory dependency, or is it an optional component?

If it is an optional component, maybe consider using a requirements specifier that is tied to a minimum python version of 3.7 using an environment marker as documented here:

https://pip.pypa.io/en/latest/reference/pip_install/#requirement-specifiers

In general, Python 3.7 is a very recent version. I see from https://pypi.org/project/pc-ble-driver-py/ that pc-ble-driver-py is also a Nordic project. Can we please add Python 3.6 support here?

@carlescufi is this something I might be able to help with?

mbolivar, 2019-10-23:
@bihanssen for context, many if not most of the NCS maintainers are using Linux, and Ubuntu 18.04 is a very popular distribution in general among them and other Linux users of Zephyr. Since NCS is using nrfutil as a dependency, I believe this issue deserves some escalation.

SebastianBoe, 2019-10-24:
I agree with Marti, we should not require 3.7 unless absolutely neccessary. It should be evaluated what it would take to make pc-ble-driver-py support 3.6.

bihanssen, 2019-10-24:
@mbolivar with regards to the misalignment of the version requirements I have updated setup.py  to make it clear that the current version requires 3.7 [#7504c2a](https://github.com/NordicSemiconductor/pc-nrfutil/commit/7504c2ac44cfb3684e5fdc784ceb6f333957495f).

@SebastianBoe the effort was already evaluated and found high to resolve issues related to 3.6 support. With the arguments raised here it could of course be taken up for reevaluation.

mbolivar, 2019-10-24:
> @mbolivar with regards to the misalignment of the version requirements I have updated setup.py to make it clear that the current version requires 3.7 [#7504c2a](https://github.com/NordicSemiconductor/pc-nrfutil/commit/7504c2ac44cfb3684e5fdc784ceb6f333957495f).

I think this patch is not correct.

PEP 440 defines `~=`, the compatible release specifier, as follows:

```
For a given release identifier V.N, the compatible release clause is
approximately equivalent to the pair of comparison clauses:

>= V.N, == V.*

This operator MUST NOT be used with a single segment version number such as ~=1.

For example, the following groups of version clauses are equivalent:

~= 2.2
>= 2.2, == 2.*
```

So `python_requires='~= 3.7'` is equivalent to `>= 3.7, == 3.*`, i.e. python 3.8 would also be considered compatible per the PEP.

I believe `python_requires= '== 3.7'` is what the semantics are, since 3.8 is not supported either, right?

bihanssen, 2019-10-25:
@mbolivar the version specifier was intentionally loose by allowing >= 3.7, ==3.* However I realize that it should rather be in sync with the requirements of pc-ble-driver-py which only works with 3.7.* 
I have updated the requirements in d16e2e2 accordingly, using the ~=3.7.0 specifier.

brainstorm, 2019-11-02:
Just to bump this up, could anybody just push HEAD to PyPi? The current version there fails with:

```shell
ERROR: Cannot uninstall 'PyYAML'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
```

OTOH, git cloned && `pip install -e .` and works great. Please save users time and confusion ;)

bihanssen, 2019-11-29:
nrfutil v6.0 final was just released. It includes support for python 3.6, 3.7 and 3.8. The version requirements are clarified in setup.py.

closed 2019-11-29 by bihanssen