NordicSemiconductor/pc-nrfutil #278: Release v6.0.0a1 crashes on Windows

cypress-speedycat, 2019-11-09:
Please use the [Nordic DevZone](https://devzone.nordicsemi.com) portal to report bugs, questions or feedback in general. If you would like to suggest code changes, feel free to submit a pull request.

cypress-speedycat, 2019-11-09:
A bit more info here. 

It's on a Windows 7 machine.

Generating keys works. The crash comes from when trying to generate a ZIP file, like `nrfutil pkg generate --debug-mode --application app.hex --key-file key.pem app_dfu_package.zip`. 

This also happens with the 5.2.0 release version.

I've tried running the `nrfutil` command in _Msys2_, _Git bash_ and _Windows Powershell_.

`pip install nrfutil` doesn't even compile with python3 on _Msys2_.

MaartenArits, 2019-11-17:
The pip package is still python 2.7, adafruit-nrfutil is python 3.5. Should pip be updated? It is not consistent with the readme.

bihanssen, 2019-12-10:
@cypress-speedycat, are seeing the issue also after release of nrfutil v6 final?

bihanssen, 2019-12-20:
Closing due to inactivity.

closed 2019-12-20 by bihanssen