NordicSemiconductor/pc-nrfutil #288: Release 6.0.0 installation failed on ubuntu 18.04

DattatrayHinge, 2019-12-06:
Hi 
I am trying to install nrfutil v6.0.0 on ubuntu 18.04 with below command and got installation error 
_**Could not find suitable distribution for Requirement.parse('pc_ble_driver_py>=0.14.0')**_

`sudo python setup.py install`

```
Processing dependencies for nrfutil==6.0.0
Searching for pc_ble_driver_py>=0.14.0
Reading https://pypi.org/simple/pc_ble_driver_py/
No local packages or working download links found for pc_ble_driver_py>=0.14.0
error: Could not find suitable distribution for Requirement.parse('pc_ble_driver_py>=0.14.0')
```

Below is details of my ubuntu machine

`Linux SVE1513CYNB 5.0.0-36-generic #39~18.04.1-Ubuntu SMP Tue Nov 12 11:09:50 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

mrquincle, 2019-12-10:
I encountered the same: https://github.com/crownstone/bluenet/issues/81

Solution: update pip

```
python3 -m pip install -U pip
```

bihanssen, 2019-12-10:
Related to https://github.com/NordicSemiconductor/pc-ble-driver-py/issues/148.

closed 2019-12-20 by bihanssen