NordicSemiconductor/pc-nrfutil #289: nrfutil 6.0.0 generates invalic c code when converting private key

mawis, 2019-12-09:
Using the new nrfutil I cannot re-generate the private key C code from the PEM file anymore.

The command I use is:

```sh
nrfutil keys display --key pk --format code ../keys/private.pem --out_file dfu_public_key.c
```

The problem is in the file `dfu_public_key.c`. The key generated there is split into two lines of 32 hex values each. The problem is that at the end of the first key line (i.e. between pk[31] and pk[32]) there is no comma. Therefore the C compiler is complaining that the file is not correct.

Generated dfu_public_key.c looks something like this:

```C
/* This file was automatically generated by nrfutil on 2019-12-09 (YY-MM-DD) at 20:21:48 */

#include "stdint.h"
#include "compiler_abstraction.h"

/** @brief Public key used to verify DFU images */
__ALIGN(4) const uint8_t pk[64] =
{
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
```

(Actual key replaced by 0x00 bytes manually by me.)

bihanssen, 2019-12-10:
@mawis, thanks for reporting, this seems like a bug.

bihanssen, 2019-12-20:
Fixed in #290

closed 2019-12-20 by bihanssen