NordicSemiconductor/pc-nrfutil #295: nrfutil v6.0.1 cannot find dylib

mrvkino, 2020-01-02:
Hi,

nrfutil v6.0.1 cannot find dylib. Version 6.0.0 works.

```bash
$nrfutil
Traceback (most recent call last):
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/bin/nrfutil", line 5, in <module>
    from nordicsemi.__main__ import cli
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/lib/python3.7/site-packages/nordicsemi/__main__.py", line 52, in <module>
    from nordicsemi.dfu.dfu_transport_serial import DfuTransportSerial
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/lib/python3.7/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 53, in <module>
    from nordicsemi.dfu.dfu_trigger import DFUTrigger
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/lib/python3.7/site-packages/nordicsemi/dfu/dfu_trigger.py", line 48, in <module>
    import usb1
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/lib/python3.7/site-packages/usb1/__init__.py", line 61, in <module>
    from . import libusb1
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/lib/python3.7/site-packages/usb1/libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "/Users/mrogersvallee/.local/share/virtualenvs/StarWars-XnlfvqiW/lib/python3.7/site-packages/usb1/libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(libusb-1.0.dylib, 6): image not found
```

bihanssen, 2020-01-03:
@mrvkino thanks for reporting. I'm not sure why nrfutil is suddenly not able to find libusb, though it seems a workaround is to install libusb with homebrew.
`brew install libusb`

bihanssen, 2020-01-03:
Seems the error is related to the introduction of an import statement:
https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/dfu_trigger.py#L48

PR #296 (2020-01-07): Fix libusb import error on macOS
This PR fixes #295 libusb import error. However it breaks libusb usage in the pyinstaller generated executable on macOS. We need to find a way to fix importing libusb later.
closed 2020-01-07 by chunfantasy