NordicSemiconductor/pc-nrfutil #307: DFU with usb-serial fails with AttributeError: 'exceptions.ImportError' object has no attribute 'strerror'

cdcformatc, 2020-03-03:
Installed nrfutil with python 3.8 and pip3 20.0.2 on Windows 10
Trying to run dfu over usb-serial:
nrfutil dfu usb-serial -pkg FW.zip -p COM4  -b 115200

I get the following error:
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 1464, in <module>
  File "click\core.py", line 764, in __call__
  File "click\core.py", line 717, in main
  File "click\core.py", line 1137, in invoke
  File "click\core.py", line 1137, in invoke
  File "click\core.py", line 956, in invoke
  File "click\core.py", line 555, in invoke
  File "nordicsemi\__main__.py", line 1001, in usb_serial
  File "nordicsemi\__main__.py", line 956, in do_serial
  File "nordicsemi\dfu\dfu.py", line 121, in dfu_send_images
  File "nordicsemi\dfu\dfu.py", line 90, in _dfu_send_image
  File "nordicsemi\dfu\dfu_transport_serial.py", line 203, in open
AttributeError: 'exceptions.ImportError' object has no attribute 'strerror'
[33728] Failed to execute script __main__

I find it a bit odd that I would be seeing an ImportError, and not an OSError or something that would indicate a serial failure. Unfortunately the bare except is keeping me from seeing what the offending import is.

5frank, 2020-03-11:
speculating it is the "dfu trigger" that raises the ImportError in combination with a badly written exception handling block:
```python
        try:
            self.__ensure_bootloader() # raises ImportError!?
            self.serial_port = Serial(...)
            self.dfu_adapter = DFUAdapter(self.serial_port)
        except Exception as e:
            ...
```

bihanssen, 2020-03-11:
@5frank, I support the theory. The exception handler assumes the exception object to have the strerror attribute, which in this case is an incorrect assumption, leading to failure. It needs a bug fix, but can be temporarily worked around by modifying the nordicsemi\dfu\dfu_transport_serial.py  file locally.

cdcformatc, 2020-03-11:
I tend to agree, regardless of the actual cause of the offending error the generic exception block assumes a specific type of exception. Would be much better written with a specific except block.

PR #319 (2020-04-24): Change to more specific exception handling in dfu transport serial
Fixes #307
closed 2020-04-24 by bihanssen