NordicSemiconductor/pc-nrfutil #309: nrfutil fails to update due to wrong name of pc-ble-driver-py

E3V3A, 2020-03-18:
Trying to update the `nrfutil` python package fails as it seem that the dependency  `pc_ble_driver_py` has changed name to `pc-ble-driver-py`...

```bash
$ pip search nrfutil
nrfutil (6.0.1)                  - Nordic Semiconductor nrfutil utility and Python library
  INSTALLED: 5.2.0
  LATEST:    6.0.1
adafruit-nrfutil (0.5.3.post12)  - Python 3 version of Nordic Semiconductor nrfutil utility and Python library (modified by Adafruit)
```

```bash
$ pip3 install --no-cache-dir --force-reinstall pc-ble-driver-py
Collecting pc-ble-driver-py
  Downloading pc_ble_driver_py-0.11.4.tar.gz (8.0 MB)
     |████████████████████████████████| 8.0 MB 1.2 MB/s
Collecting wrapt
  Downloading wrapt-1.12.1.tar.gz (27 kB)
Collecting future
  Downloading future-0.18.2.tar.gz (829 kB)
     |████████████████████████████████| 829 kB 1.2 MB/s
Building wheels for collected packages: pc-ble-driver-py, wrapt, future
  Building wheel for pc-ble-driver-py (setup.py) ... done
  Created wheel for pc-ble-driver-py: filename=pc_ble_driver_py-0.11.4-py3-none-any.whl size=8095779 sha256=a166375bcc14497d620079e09a43768009179e23050bf69d22a18989a4bcf4da
  Stored in directory: /tmp/pip-ephem-wheel-cache-fudgqbn9/wheels/39/f5/9a/55f39f0f091a0f6c315e332ec1c909c52e2819bc2972c2494a
  Building wheel for wrapt (setup.py) ... done
  Created wheel for wrapt: filename=wrapt-1.12.1-py3-none-any.whl size=19553 sha256=cd038daef74fd1b8f6852790de01e003f4a65904a8f4dadd4f025a9966b50c91
  Stored in directory: /tmp/pip-ephem-wheel-cache-fudgqbn9/wheels/5f/fd/9e/b6cf5890494cb8ef0b5eaff72e5d55a70fb56316007d6dfe73
  Building wheel for future (setup.py) ... done
  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491058 sha256=021cf87b96796b88d64834e24ff7d84054b1f6756748372558d75ffe93c46643
  Stored in directory: /tmp/pip-ephem-wheel-cache-fudgqbn9/wheels/8e/70/28/3d6ccd6e315f65f245da085482a2e1c7d14b90b30f239e2cf4
Successfully built pc-ble-driver-py wrapt future
ERROR: nrfutil 5.2.0 has requirement ecdsa~=0.13.0, but you'll have ecdsa 0.15 which is incompatible.
ERROR: nrfutil 5.2.0 has requirement piccata~=1.0, but you'll have piccata 2.0.0 which is incompatible.
ERROR: nrfutil 5.2.0 has requirement pyyaml~=4.2b1, but you'll have pyyaml 5.1.2 which is incompatible.
Installing collected packages: wrapt, future, pc-ble-driver-py
  Attempting uninstall: wrapt
    Found existing installation: wrapt 1.11.2
    Uninstalling wrapt-1.11.2:
      Successfully uninstalled wrapt-1.11.2
  Attempting uninstall: future
    Found existing installation: future 0.18.2
    Uninstalling future-0.18.2:
      Successfully uninstalled future-0.18.2
  WARNING: The scripts futurize and pasteurize are installed in '/cygdrive/c/cygwin64/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  Attempting uninstall: pc-ble-driver-py
    Found existing installation: pc-ble-driver-py 0.11.4
    Uninstalling pc-ble-driver-py-0.11.4:
      Successfully uninstalled pc-ble-driver-py-0.11.4
Successfully installed future-0.18.2 pc-ble-driver-py-0.11.4 wrapt-1.12.1

$ pip3 install --no-cache-dir --force-reinstall nrfutil
Collecting nrfutil
  Downloading nrfutil-6.0.1.tar.gz (843 kB)
     |████████████████████████████████| 843 kB 829 kB/s
Collecting click
  Downloading click-7.1.1-py2.py3-none-any.whl (82 kB)
     |████████████████████████████████| 82 kB 1.1 MB/s
Collecting crcmod
  Downloading crcmod-1.7.tar.gz (89 kB)
     |████████████████████████████████| 89 kB 1.1 MB/s
Collecting ecdsa
  Downloading ecdsa-0.15-py2.py3-none-any.whl (100 kB)
     |████████████████████████████████| 100 kB 1.2 MB/s
Collecting intelhex
  Downloading intelhex-2.2.1-py2.py3-none-any.whl (50 kB)
     |████████████████████████████████| 50 kB 1.3 MB/s
Collecting libusb1
  Downloading libusb1-1.7.1.tar.gz (78 kB)
     |████████████████████████████████| 78 kB 1.4 MB/s
ERROR: Could not find a version that satisfies the requirement pc_ble_driver_py>=0.14.1 (from nrfutil) (from versions: 0.1.0, 0.2.0, 0.3.0, 0.4.0, 0.5.0, 0.6.0, 0.6.1, 0.6.2, 0.8.0, 0.8.1, 0.9.0, 0.9.1, 0.10.0, 0.11.0, 0.11.1, 0.11.2, 0.11.3, 0.11.4)
ERROR: No matching distribution found for pc_ble_driver_py>=0.14.1 (from nrfutil)
```

E3V3A, 2020-03-18:
or maybe something else?

bihanssen, 2020-03-18:
What is the pip version? Required pip version is >= 19.0.

bihanssen, 2020-04-07:
I believe this is due to pip version as mentioned. Please reopen if this is not the case.

closed 2020-04-07 by bihanssen