NordicSemiconductor/pc-nrfutil #317: Support for multiple SDK versions

mrquincle, 2020-04-10:
It would be great to have support for multiple SDK versions.

For example [here](https://devzone.nordicsemi.com/f/nordic-q-a/41650/nrf51822-nrfutil-dfu-ble) it stated that you need:

* Version 0.5.2 generates legacy firmware packages compatible with nRF SDK 11.0 and older
* Versions 1.5.0 and later generate modern firmware packages compatible with nRF SDK 12.0 and newer
* Versions 4.0.0 and later generate modern firmware packages compatible with nRF SDK 15.1 and newer

You have implemented the different versions over time, so it would be great to have a command line option that selects one of those three formats. Then people do not have to have multiple of those tools on their system depending on the version with which the particular firmware was created.

More importantly, e.g. nrfutil-0.5.2 does not support DFU over BLE. It would be great to be able to perform DFU from the PC to all Nordic boards we have.

ERiSugawara, 2020-06-08:
Products using SDK11 need to be supported for the long term yet. However, nrfutil-0.5.x is built on the deprecated Python 2.

This proposal will also benefit manufacturers using such hardware components.

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr