NordicSemiconductor/pc-nrfutil #331: TypeError in 'nrfutil pkg generate' - Provide useful feedback

epac-tom, 2020-08-13:
Following the Getting started with Nordic's Secure DFU bootloader guide. 
When running nrfutil to generate the DFU package using the nrfutil pkg generate command it will bomb out with a TypeError, generated here: https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/nrfhex.py#L135
Some searching revealed this post: https://devzone.nordicsemi.com/f/nordic-q-a/48245/typeerror-in-nrfutil-pkg-generate
Indicating the issue with using a project without the correct offset.

Can the code be modified to generate a useful warning message? I'm sure I won't be the last new user to run into this issue, a quick warning would be very useful instead of having to google and eventually find the above post before realising the mistake made.

bihanssen, 2020-11-24:
@epac-tom, thanks for the input. To be evaluated for next release.

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr