NordicSemiconductor/pc-nrfutil #335: DFU HW SD INFO

cederom, 2020-11-29:
Hello world :-)

Is it possible to display some information on currently connected DFU devices with `nrfutil`?

I am using nRF52840 DONGLE and `nrfutil` on FreeBSD OS in order to sniff some Bluetooth comms. The Sniffer package only contains HEX file and instructions to use some desktop application which does not work here. I can only use `nrfutil`.

I need to generate DFU ZIP package on my own. I need to provide HW and SD information which I do not know. It would be really nice if `nrfutil` could tell me some details about currently connected DFU target so I can know what numbers I can provide in order to generate that DFU ZIP.

Any hints welcome :-)
Tomek

cederom, 2020-11-29:
Also I do not know the key to be used in order to sign the DFU ZIP package.. but I think I managed to skip that part using `--debug-mode` switch to `pkg generate` :-)

cederom, 2020-11-29:
It seems I made the package with

```
nrfutil  pkg generate --debug-mode --application sniffer_nrf52840dongle_nrf52840_7cc811f.hex --hw-version 52 --sd-req 0 sniffer_nrf52840dongle_nrf52840_7cc811f.zip
Failed to import ecdsa, cannot do signing

|===============================================================|
|##      ##    ###    ########  ##    ## #### ##    ##  ######  |
|##  ##  ##   ## ##   ##     ## ###   ##  ##  ###   ## ##    ## |
|##  ##  ##  ##   ##  ##     ## ####  ##  ##  ####  ## ##       |
|##  ##  ## ##     ## ########  ## ## ##  ##  ## ## ## ##   ####|
|##  ##  ## ######### ##   ##   ##  ####  ##  ##  #### ##    ## |
|##  ##  ## ##     ## ##    ##  ##   ###  ##  ##   ### ##    ## |
| ###  ###  ##     ## ##     ## ##    ## #### ##    ##  ######  |
|===============================================================|
|You are generating a package with the debug bit enabled in the |
|init packet. This is only compatible with a debug bootloader   |
|and is not suitable for production.                            |
|===============================================================|

|===============================================================|
|##      ##    ###    ########  ##    ## #### ##    ##  ######  |
|##  ##  ##   ## ##   ##     ## ###   ##  ##  ###   ## ##    ## |
|##  ##  ##  ##   ##  ##     ## ####  ##  ##  ####  ## ##       |
|##  ##  ## ##     ## ########  ## ## ##  ##  ## ## ## ##   ####|
|##  ##  ## ######### ##   ##   ##  ####  ##  ##  #### ##    ## |
|##  ##  ## ##     ## ##    ##  ##   ###  ##  ##   ### ##    ## |
| ###  ###  ##     ## ##     ## ##    ## #### ##    ##  ######  |
|===============================================================|
|You are not providing a signature key, which means the DFU     |
|files will not be signed, and are vulnerable to tampering.     |
|This is only compatible with a signature-less bootloader and is|
|not suitable for production environments.                      |
|===============================================================|

Zip created at sniffer_nrf52840dongle_nrf52840_7cc811f.zip
```

Flashed it with 

```
 nrfutil --verbose dfu usb-serial -p /dev/cuaU0 -pkg sniffer_nrf52840dongle_nrf52840_7cc811f.zip
Failed to import ecdsa, cannot do signing
  [####################################]  100%
Device programmed.
```

As a result I got umodem device up and running :-)

```
ugen0.12: <ZEPHYR nRF Sniffer for Bluetooth LE> at usbus0
umodem0 on uhub0
umodem0: <ZEPHYR nRF Sniffer for Bluetooth LE, class 0/0, rev 2.00/2.03, addr 30> on usbus0
umodem0: data interface 1, has CM over data, has no break
```

The OS is FreeBSD :-)

```
FreeBSD 0xCFMX4 12.2-RELEASE FreeBSD 12.2-RELEASE r366954 GENERIC  amd64
```

cederom, 2020-11-29:
The question left is: can `nrfutil` provide some useful information about the target DFU device? That would be very handy here because no other utilities seems available :-)

bihanssen, 2021-01-13:
@cederom, sorry but there are no current plans to add such a feature. You are free to provide a PR if you have a solution in mind.

closed 2021-01-13 by bihanssen