NordicSemiconductor/pc-nrfutil #336: python 3.9 / pip 29.3.1 installs 5.2.0 instead of 6.1.0 on macos

myst729, 2020-12-09:
Even if with `--no-cache-dir`.

libcg, 2021-01-12:
Same issue on Arch Linux with `pip3`.

libcg, 2021-01-12:
Fixed it by running the following command after `pip3 install nrfutil` (Arch only):
`# pip3 install --no-deps -t /usr/lib/python3.9/site-packages --python-version 3.8 --ignore-requires-python --upgrade nrfutil`

myst729, 2021-01-12:
Workaround here: https://github.com/ferrous-systems/embedded-trainings-2020/pull/115/

bihanssen, 2021-01-13:
Thanks for the info @libcg. I suspect you might run into problems if running commands that actually require ble-driver-py, like DFU over BLE.
Install fails because ble-driver-py does not yet include wheels for python versions other than 3.6-3.8.

myst729, 2021-01-14:
> Thanks for the info @libcg. I suspect you might run into problems if running commands that actually require ble-driver-py, like DFU over BLE.
> Install fails because ble-driver-py does not yet include wheels for python versions other than 3.6-3.8.

I did. So it looks like nrfutil is waiting for *ble-driver-py* to be compatible with python 3.9

chrysn, 2021-02-11:
If there are dependencies that have a habit of breaking with compatible Python versions, could they become optional? There is the "extras" mechanism that'd allow just that.

bihanssen, 2021-02-11:
@chrysn that would be useful. Have tried similar in the past without success. If you have a PR to demonstrate that would be helpful.

chrysn, 2021-02-11:
Sorry, can't help with a PR here for policy reasons (as this module is not published under an OSI-approved license).

gelintonx, 2021-06-27:
Same problem with Parrot Os and Python 3.9, I had to downgrade to python 8 to get the latest version

myst729, 2021-08-29:
6.1.2 works

closed 2021-08-29 by myst729