NordicSemiconductor/pc-nrfutil #337: Current nrfutil 6.1.0 is not installable on Arm-64 architectures via pip

vollbier, 2020-12-09:
Hi,
I tried installing nrfutil with `pip3 install nrfutil` onto a Raspberry Pi 3 with Debian buster on it (`Linux my-rpi3 4.19.0-13-arm64 #1 SMP Debian 4.19.160-2 (2020-11-28) aarch64 GNU/Linux`), which did not work due to dependency problems. Note that

* calling `pip3 install  pc_ble_driver_py` installs version `0.11.4` instead of `0.14.2`  and
* installing nrfutil 6.0.1 with `pip3 install --no-dependencies nrfutil==6.0.1` after installing the required dependencies manually worked before.

Thanks, dietrich

PS: This is the output when trying to the current rftutil:

    $ pip3 install nrfutil
    Collecting nrfutil
    Downloading https://files.pythonhosted.org/packages/c6/17/4c07a22099b92ab56a7f511080b815d804b7441382c5c0f821960f05d2dd/nrfutil-6.1.0.tar.gz (842kB)
        100% |████████████████████████████████| 849kB 98kB/s 
    Collecting click (from nrfutil)
    Downloading https://files.pythonhosted.org/packages/d2/3d/fa76db83bf75c4f8d338c2fd15c8d33fdd7ad23a9b5e57eb6c5de26b430e/click-7.1.2-py2.py3-none-any.whl (82kB)
        100% |████████████████████████████████| 92kB 957kB/s 
    Collecting crcmod (from nrfutil)
    Downloading https://files.pythonhosted.org/packages/6b/b0/e595ce2a2527e169c3bcd6c33d2473c1918e0b7f6826a043ca1245dd4e5b/crcmod-1.7.tar.gz (89kB)
        100% |████████████████████████████████| 92kB 913kB/s 
    Collecting ecdsa (from nrfutil)
    Downloading https://files.pythonhosted.org/packages/98/16/70be2716e24eaf5d81074bb3c05429d60292c2a96613a78ac3d69526ad2a/ecdsa-0.16.1-py2.py3-none-any.whl (104kB)
        100% |████████████████████████████████| 112kB 893kB/s 
    Collecting intelhex (from nrfutil)
    Downloading https://files.pythonhosted.org/packages/97/78/79461288da2b13ed0a13deb65c4ad1428acb674b95278fa9abf1cefe62a2/intelhex-2.3.0-py2.py3-none-any.whl (50kB)
        100% |████████████████████████████████| 51kB 980kB/s 
    Collecting libusb1 (from nrfutil)
    Downloading https://files.pythonhosted.org/packages/3c/4b/0c8b566d21f8920019237cc9f919e0aeeb43aa350819ea4bcdd2788f39a3/libusb1-1.9.1-py3-none-any.whl (58kB)
        100% |████████████████████████████████| 61kB 102kB/s 
    Collecting pc_ble_driver_py>=0.14.2 (from nrfutil)
    Could not find a version that satisfies the requirement pc_ble_driver_py>=0.14.2 (from nrfutil) (from versions: 0.1.0, 0.2.0, 0.3.0, 0.4.0, 0.5.0, 0.6.0, 0.6.1, 0.6.2, 0.8.0, 0.8.1, 0.9.0, 0.9.1, 0.10.0, 0.11.0, 0.11.1, 0.11.2, 0.11.3, 0.11.4)
    No matching distribution found for pc_ble_driver_py>=0.14.2 (from nrfutil)

bihanssen, 2021-01-13:
Hi, pc-ble-driver-py currently only supports python 3.6-3.8. Which versions of python and pip are you using?

bihanssen, 2021-01-13:
.. and arm is not supported. ble-driver-py does not contain wheels compiled for arm architecture.

vollbier, 2021-01-14:
I tried it with Python 3.7. As already noted, a colleague of mine claimed that he got it running some time ago with the following approach:

        apt-get -y install libusb-1.0-0-dev python3 python3-pip sed
        pip3 install click crcmod ecdsa intelhex libusb1 piccata protobuf
        pip3 install pyserial pyyaml tqdm pc_ble_driver_py pyspinel
        pip3 install --no-dependencies nrfutil==6.0.1

There, version 6.0.1 is used.

amunozh, 2021-03-29:
Hello, I've been struggling with the same problem, I was trying to install nrfutil in a Raspberry Pi 3 with Raspbian Stretch but It have a lot of dependencies, so I decided to make a container for arm32v7 architectures.  If you still looking for a solution you can pull it from my repository `amunoz20/nrf52-dongle:latest`
Otherwise, this is the Dockerfile that I did (I know it is not too much optimized, but It is working, I just tested it for flashing another device using BLE). I hope it will be useful for you or for another searching how to run a nrfutil in arm architectures.

```
FROM arm32v7/python:3.7-slim-buster

WORKDIR /app

COPY aptconf.txt /etc/apt/apt.conf

RUN apt-get update

RUN apt-get install build-essential git autoconf libudev-dev ninja-build libboost-dev wget cmake swig make gcc g++ libspdlog-dev libasio-dev libfmt-dev -y

RUN git clone --branch v2.13.2 https://github.com/catchorg/Catch2.git

WORKDIR Catch2

RUN cmake -Bbuild -H. -DBUILD_TESTING=OFF && cmake --build build/ --target install

WORKDIR /app

RUN git clone https://github.com/NordicSemiconductor/pc-ble-driver.git

WORKDIR pc-ble-driver

RUN mkdir build && cd build && cmake -G Ninja -DNRF_BLE_DRIVER_VERSION=4.1.4 -DCMAKE_BUILD_TYPE=Release .. && cmake --build . --target install

WORKDIR /app

RUN git clone https://github.com/NordicSemiconductor/pc-ble-driver-py.git

COPY requirements.txt ./

RUN pip install --upgrade pip

RUN pip install -r requirements.txt

WORKDIR pc-ble-driver-py

RUN python setup.py bdist_wheel --build-type Release

RUN apt-get install libffi-dev curl

#RUN curl https://sh.rustup.rs -sSf | bash -s -- -y

#ENV PATH="/root/.cargo/bin:${PATH}"

ENV CRYPTOGRAPHY_DONT_BUILD_RUST=1

RUN pip install cryptography

RUN pip install wrapt

RUN pip install --no-index --find-links=file:///app/pc-ble-driver-py/dist/pc_ble_driver_py-0.16.0-cp37-cp37m-linux_armv7l.whl pc_ble_driver_py==0.16.0

WORKDIR /app

RUN git clone https://github.com/NordicSemiconductor/pc-nrfutil.git

WORKDIR pc-nrfutil

RUN python setup.py install

USER root

COPY . .
```

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr