NordicSemiconductor/pc-nrfutil #339: Installation fails: ImportError: No module named setuptools_behave

NanaDada, 2020-12-22:
Calling `sudo python setup.py install` (on Ubuntu) fails with:
```
Traceback (most recent call last):
  File "setup.py", line 44, in <module>
    from setuptools_behave import behave_test
ImportError: No module named setuptools_behave

```
Do you have any ideas how to fix this?

bihanssen, 2021-01-13:
@NanaDada, maybe if you try to install requirements-dev with `pip install -r requirements-dev.txt` first, as it includes the `behave`module.

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr