NordicSemiconductor/pc-nrfutil #342: Update --sd-req (list of SoftDevice firmware IDs, FWID) aligned to latest SDK 17.0.2

chrislro2019, 2021-01-08:
E.g. related to DevZone Case ID: 262105: FWID for s132 v7.2.0 is 0x0101.
See https://devzone.nordicsemi.com/f/nordic-q-a/69907/bootloader-dfu-buttonless/286737

chrislro2019, 2021-07-06:
See also https://devzone.nordicsemi.com/f/nordic-q-a/76941/what-should-the-value-of--sd--req-be-to-fit-s132_nrf52_7-2-0_softdevice/ 
When updating you should also respect latest SDs i.e. 7.3.0 of S112, S113, S132, S140 (as announced mid of 2021)

chrislro2019, 2021-09-13:
As the FWIDs are still not in 6.1.2 here are the FWID numbers from the SD release note PDF:
- S132 v7.3.0 = 0x0124
- S132 v7.2.1 = 0x0117
- S132 v7.2.0 = 0x0101

Would be nice to add the latest ones to the "non-exhaustive" list in https://github.com/NordicSemiconductor/pc-nrfutil/blob/10276bfb40b45a55722d48c908b70457f717ceba/nordicsemi/__main__.py#L529 instead of the old values from the past.

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr