NordicSemiconductor/pc-nrfutil #347: how can I use mesh dfu nrfutil in python 3.7 with nrfutil 6.1

Sama-L, 2021-02-16:
Please use the [Nordic DevZone](https://devzone.nordicsemi.com) portal to report bugs, questions or feedback in general. If you would like to suggest code changes, feel free to submit a pull request.
I am going to run the DFU over Mesh. When I used this command (nrfutil dfu serial -pkg dfu_test.zip -p <COM port> -b 115200 -fc --mesh) I faced error containing : 
invalid value for -fc: --mesh is not a valid boolean

How can I use this command?

Thank you,
Sama

peterhinson, 2021-02-21:
@Sama-L I believe you'll need to use the [mesh_dfu](https://github.com/NordicSemiconductor/pc-nrfutil/tree/mesh_dfu) branch. Unfortunately, it's painfully out of date, requires Python 2.7, and presents a name conflict with the modern nrfutil. 

Checkout [DFU Configuration Requirements](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.meshsdk.v5.0.0/md_doc_user_guide_modules_dfu_configuring_performing.html#dfu_configuration_requirements) for more details.

Fyi, in the command you posted, you'll want to pass a port for `-p` and a bool for `-fc` (e.g., `-p /dev/ttyyourboard -fc true`).

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr