NordicSemiconductor/pc-nrfutil #358: Release 6.1 (or 6.1.1) not fully available anymore?

rabarberski, 2021-08-24:
nrfutil-linux used to be available on github for release 6.1 under 
https://github.com/NordicSemiconductor/pc-nrfutil/releases (and, more specifically, under https://github.com/NordicSemiconductor/pc-nrfutil/releases/download/v6.1/nrfutil-linux)

On github, only the source of 6.1 is available

Have the 6.1 assets been removed from github as a release package?

I also noticed there is an official 6.1 and 6.1.1 release via pypi (https://pypi.org/project/nrfutil/#history), so it's doubly weird it's not on github

jornbh, 2021-08-24:
I am not sure what may have happened to the 6.1 release on GitHub for version 6.1, but we may look into it if there is a specific need for it. 

You are absolutely tight in that there has been a release on pip. A proper GitHub release for version 6.1.1 has now been added, which should hopefully solve your problem. 

If you need version 6.1 specifically, you simply ask.

rabarberski, 2021-08-24:
Thx for adding this so quickly. 
No need (for me) for the 6.1, although it did break our CI/CD chain (was fetching it via wget). I switched to pip install already, as I hadn't expected the fix so fast.

closed 2021-08-24 by rabarberski