NordicSemiconductor/pc-nrfutil #364: how can I use s140_nrf52_7.2.0_softdevice with nrfutil 6.1.3

naluckstar, 2021-11-05:
When will a new version be released and compatible with S140_ nrf52_ 7.2.0_ softdevice？ Current version, up to s140_nrf52_7.0.1, but S140_ nrf52_ 7.0.1. i can't find the download link... With the old version, there is a problem with the Bluetooth connection process. s140_nrf52_7.2.0_softdevice version can solve the problem. Please help solve this embarrassing problem.Or is there any other way to help generate the final burning file

happytaoxiaoli, 2022-01-18:
确定不适配吗？ 
我也是nrfutil 6.1.3 和 s140_nrf52_7.2.0_softdevice，用来制作ota升级包，没什么问题，只要使用正确的softdevice id就可以了。
S140 v7.2.0 的softdevice id 是0x100

chunfantasy, 2022-01-18:
@happytaoxiaoli appreciated for your answer. Would be nice to give the answer in English so more people could understand ❤️

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

The SoftDevice v6.X and higher will not be supported in this utility or the new one with the same name.

closed 2023-01-17 by kenr