NordicSemiconductor/pc-nrfutil #366: nrfutil not compatible with python 3.10 or later release

nullr0ute, 2021-11-22:
When trying to pip install nrfutil with python 3.10 or later release it installs the 5.1.0 release which isn't compatible with python3 so it doesn't work.

GaryDT, 2021-12-23:
I'm also using python 3.10 and having problems. After running the install process via pip3 in a command window, it seems to have  completed ok, but when I try to do anything such as nrfutil --help, I get various traceback errors.

I had a similar problem when trying to install Zephyr about 7 weeks ago, which incidentally I never completed due to getting a little side-tracked. I'll be trying to install nRF Util again this evening via python 3.9

Here's the Zephyr link just in case anyone's interested... [PyPI Wheels for Python 3.10](https://github.com/zephyrproject-rtos/windows-curses/issues/26)

GaryDT, 2021-12-23:
I'm using Windows and also using Python 3.10 and having problems. After running the install process via pip3 in a command window, it seems to have  completed ok, but when I try to do anything such as nrfutil --help, I get various traceback errors.

I had a similar problem when trying to install Zephyr about 7 weeks ago, which incidentally I never completed due to getting a little side-tracked. I'll be trying to install nRF Util again this evening via Python 3.9

It's not related to the Windows system path either, which I know has been added correctly, and also nRF Util has indeed been installed into the "C:\Users\<username>\AppData..." correctly since I can see the .exe

Here's the Zephyr link just in case anyone's interested... [PyPI Wheels for Python 3.10](https://github.com/zephyrproject-rtos/windows-curses/issues/26)

**Edit:** maybe it's because I was installing via the command line that I missed the following...

[nrfutil 6.1.3](https://pypi.org/project/nrfutil/)

>Requires: Python >=3.7, <3.10

**Edit 2:** yes indeed it states 3.7 or later (see image below), which contradicts the above.

[image]

**Edit 3:** Python 3.9.9 worked very nicely :) scroll to the very bottom of the following page to find [Python "Windows installer (64-bit)" 3.9.9](https://www.python.org/downloads/release/python-399/) ... (**I've tested "nrfutil --help" as below**)

[image]

andreaaizza, 2022-05-19:
Hi Nordic, this is 6 months old. We cannot run nrf-util on Python 3.10. When will this be fixed?

mrquincle, 2022-06-01:
This repository becomes more and more outdated.

I can recommend people to use `pyenv` in combination with `direnv` to be able to cope with slow-moving repositories like these. This is the contents of the `.envrc` file in my working directory.

```
layout pyenv 3.9.13
echo "Hold back to python 3.9.13 because of nrfutil"
echo "See https://github.com/NordicSemiconductor/pc-nrfutil/issues/366"
```

Note that you will probably also run in the lax requirements for protobuf as e.g. described in another issue: https://github.com/NordicSemiconductor/pc-nrfutil/issues/350

```
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).
```

You can just limit the version for this project.

```
pip install protobuf==4.21.1
```

I've take a bit of time to explain it because these kinds of "tricks" to properly set up your python environment might be a bit more foreign to embedded developers. I'll hopefully give someone the tools rather than just the solution in this way. :-)

towen, 2022-06-22:
I found that "it works" without a venv if I use `pip install --ignore-requires-python`

ohadshapira, 2023-01-02:
2023 is here, any update regarding this issue?

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr