NordicSemiconductor/pc-nrfutil #368: Installing nrfutil using pip, but installed nrfutil-5.2.0

HeFeng1947, 2021-11-23:
I use ubuntu 20.04 server and output log as follow:
```shell
ubuntu@ubuntu:~/workspace/otbr$ python3 -V
Python 3.8.10
ubuntu@ubuntu:~/workspace/otbr$ 
ubuntu@ubuntu:~/workspace/otbr$ 
ubuntu@ubuntu:~/workspace/otbr$ pip install nrfutil
Defaulting to user installation because normal site-packages is not writeable
Collecting nrfutil
  Using cached nrfutil-6.1.3.tar.gz (844 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: click in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (7.1.2)
Requirement already satisfied: crcmod in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.7)
Requirement already satisfied: ecdsa in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (0.13.3)
Requirement already satisfied: intelhex in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (2.3.0)
Requirement already satisfied: libusb1==1.9.3 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.9.3)
  Using cached nrfutil-6.1.2.tar.gz (844 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-6.1.1.tar.gz (841 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-6.1.0.tar.gz (842 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-6.0.1.tar.gz (843 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-5.2.0-py3-none-any.whl
Requirement already satisfied: enum34~=1.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.1.10)
Requirement already satisfied: pc-ble-driver-py~=0.11.4 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (0.11.4)
Requirement already satisfied: pyspinel==1.0.0a3 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.0.0a3)
Requirement already satisfied: piccata~=1.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.0.1)
Requirement already satisfied: pyyaml~=4.2b1 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (4.2b4)
Requirement already satisfied: tqdm~=4.25 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (4.62.3)
Requirement already satisfied: pyserial~=3.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (3.5)
Requirement already satisfied: six~=1.9 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.16.0)
Requirement already satisfied: behave~=1.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.2.6)
Requirement already satisfied: protobuf~=3.6 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (3.19.1)
Requirement already satisfied: ipaddress in /home/ubuntu/.local/lib/python3.8/site-packages (from pyspinel==1.0.0a3->nrfutil) (1.0.23)
Requirement already satisfied: parse>=1.8.2 in /home/ubuntu/.local/lib/python3.8/site-packages (from behave~=1.0->nrfutil) (1.19.0)
Requirement already satisfied: parse-type>=0.4.2 in /home/ubuntu/.local/lib/python3.8/site-packages (from behave~=1.0->nrfutil) (0.5.2)
Requirement already satisfied: future in /home/ubuntu/.local/lib/python3.8/site-packages (from pc-ble-driver-py~=0.11.4->nrfutil) (0.18.2)
Requirement already satisfied: wrapt in /home/ubuntu/.local/lib/python3.8/site-packages (from pc-ble-driver-py~=0.11.4->nrfutil) (1.13.3)
Installing collected packages: nrfutil
Successfully installed nrfutil-5.2.0
```

I don't know what happened, why was the last NRFUtil installed version 5.2.0?

vidarbe, 2021-11-23:
Please upgrade pip and try again.
$ pip install --user -U pip
$ pip install --user -U nrfutil

HeFeng1947, 2021-11-24:
> pip install --user -U nrfutil

Thank you for your reply!
Unfortunately, it doesn't work.
```shell
ubuntu@ubuntu:~/workspace/otbr/pc-nrfutil$ pip install --user -U pip
Requirement already satisfied: pip in /home/ubuntu/.local/lib/python3.8/site-packages (21.3.1)
ubuntu@ubuntu:~/workspace/otbr/pc-nrfutil$ pip install --user -U nrfutil
Requirement already satisfied: nrfutil in /home/ubuntu/.local/lib/python3.8/site-packages (5.2.0)
Collecting nrfutil
  Using cached nrfutil-6.1.3.tar.gz (844 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: click in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (7.1.2)
Requirement already satisfied: crcmod in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.7)
Requirement already satisfied: ecdsa in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (0.13.3)
Requirement already satisfied: intelhex in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (2.3.0)
Requirement already satisfied: libusb1==1.9.3 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.9.3)
  Using cached nrfutil-6.1.2.tar.gz (844 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-6.1.1.tar.gz (841 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-6.1.0.tar.gz (842 kB)
  Preparing metadata (setup.py) ... done
  Using cached nrfutil-6.0.1.tar.gz (843 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: pyspinel==1.0.0a3 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.0.0a3)
Requirement already satisfied: pyserial~=3.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (3.5)
Requirement already satisfied: pyyaml~=4.2b1 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (4.2b4)
Requirement already satisfied: enum34~=1.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.1.10)
Requirement already satisfied: pc-ble-driver-py~=0.11.4 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (0.11.4)
Requirement already satisfied: piccata~=1.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.0.1)
Requirement already satisfied: six~=1.9 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.16.0)
Requirement already satisfied: tqdm~=4.25 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (4.62.3)
Requirement already satisfied: behave~=1.0 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (1.2.6)
Requirement already satisfied: protobuf~=3.6 in /home/ubuntu/.local/lib/python3.8/site-packages (from nrfutil) (3.19.1)
Requirement already satisfied: ipaddress in /home/ubuntu/.local/lib/python3.8/site-packages (from pyspinel==1.0.0a3->nrfutil) (1.0.23)
Requirement already satisfied: parse>=1.8.2 in /home/ubuntu/.local/lib/python3.8/site-packages (from behave~=1.0->nrfutil) (1.19.0)
Requirement already satisfied: parse-type>=0.4.2 in /home/ubuntu/.local/lib/python3.8/site-packages (from behave~=1.0->nrfutil) (0.5.2)
Requirement already satisfied: wrapt in /home/ubuntu/.local/lib/python3.8/site-packages (from pc-ble-driver-py~=0.11.4->nrfutil) (1.13.3)
Requirement already satisfied: future in /home/ubuntu/.local/lib/python3.8/site-packages (from pc-ble-driver-py~=0.11.4->nrfutil) (0.18.2)
ubuntu@ubuntu:~/workspace/otbr/pc-nrfutil$ nrfutil version
nrfutil version 5.2.0
```

vidarbe, 2021-11-24:
Strange, I'm running ubuntu 20.04 with the same pip version myself, and I'm unable to replicate the issue here. Can you please try to uninstall the existing version of pc-ble-driver-py and nrfutil before installing nrfutil again?

gudnimg, 2022-01-06:
I had this issue on Python 3.10.1, I uninstalled it and installed Python 3.9.9, then it worked. See the version here: https://pypi.org/project/nrfutil/ 
> Requires: Python >=3.7, <3.10

sl8vz, 2022-04-19:
Nordic please update pip packages for more python version compatibility.

nullr0ute, 2022-10-18:
So this was fixed to work with pyton 3.10 but now doesn't work with 3.11 :-1:

nullr0ute, 2022-10-18:
Duplicate of #366

kenr, 2023-01-17:
Hi! We do not have capacity to maintain and release new versions of this utility.
We have made a new utility with the same name and feature set. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr