NordicSemiconductor/pc-nrfutil #391: protobuf bug causes sd-only package generation to fail on python win32

mickael9, 2022-10-25:
I was trying to generate a DFU package with just the softdevice, like this:

```
nrfutil pkg generate --key-file key.pem --hw-version 52 --sd-req 0xC1 --softdevice s140_nrf52_7.0.1_softdevice.hex out.zip
```

And getting the following error:
```
Traceback (most recent call last):
  ...
  File "...\lib\site-packages\nordicsemi\dfu\init_packet_pb.py", line 142, in _validate
    raise RuntimeError("Invalid range of firmware argument. [0 - 0xffffffff] is valid range")
RuntimeError: Invalid range of firmware argument. [0 - 0xffffffff] is valid range
```

It seems that the protobuf-generated `InitCommand()` class will interpret 0xFFFFFFFF as -1, as can be demonstrated with the following code:
```python
import dfu_cc_pb2 as pb

p = pb.InitCommand(fw_version=0xffffffff)
print(p.fw_version) # can print -1 instead of 4294967295
```

Only **32-bit** Python on Windows seems to be affected (tested with Python 3.7 & 3.10 Win32, don't know if it happens on Linux 32)
The bug was introduced in protobuf 3.19 (downgrading using `pip install protobuf<3.19` fixes the issue)

I suppose this could be fixed by either modifying the `_validate` logic or specifying `protobuf<3.19` in requirements.
Alternatively, upgrading to protobuf 4 seems to resolve the issue (but `dfu_cc_pb2.py` needs to be regenerated)

nunof, 2022-11-20:
Also happens in Linux 64bits (Ubuntu 22.10). PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python is simple workaround but upgrading to protobuf 4 would be nice.

kenr, 2023-01-17:
We do not have capacity to maintain and release new versions of this utility.
A new utility with the same name will cover this feature. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr